hku-mars/HBA #3: rviz no data

jkff00, 2023-06-12:
I tried the park data and it does listen to some relevant topics, but rviz doesn't show anything (nor any output files at the end), what could be the problem?
[image]
[image]

samsdolphin, 2023-06-12:
Hello, I forget to mention, the `hba.launch` only optimizes the poses but does not visualize the point cloud. You could view the point cloud using `visualize.launch`. Hope this helps.

jkff00, 2023-06-12:
yep,i try and it's work.

WINNER 东区
***@***.***

抄送: "WINNER ***@***.******@***.***&gt;;

closed 2023-06-12 by samsdolphin

Fury-k5, 2025-03-28:
I have the same problem as you, but after I launch visualize.launch, the rviz interface still doesn't show anything, do you know what's the reason for this?