hku-mars/HBA #5: I think the code and the contents of the paper are different.

jaeseok4104, 2023-06-15:
The code has a constraint between all the nodes in the window, and didn't make the constraint of the floor 2 layer.

Can you explain this?

[image]

[image]

samsdolphin, 2023-06-16:
Hi @jaeseok4104, yes you are right. In short, we found this implementation has a faster convergence rate and a better precision. We thus open source this version of code instead of the very original one (paper version). However, you could modify the code structure by yourself. Hope this helps.

jaeseok4104, 2023-06-16:
thanks :) Have you plan to open about paper version code?

samsdolphin, 2023-06-16:
Haha, this shouldn't be hard for you ;)

closed 2023-06-16 by samsdolphin

HaisenbergPeng, 2023-08-07:
Hi, after the sliding window ba in the 1st layer， the optimized poses are not used for factor graph, instead, you used the initial poses and the ba hessians. Could you please explain the reason? thanks a lot!
[image]

[image]

samsdolphin, 2023-08-07:
Hi @HaisenbergPeng, in each local ba, we want the hessians to constrain the relative poses between the 1st layer poses. The optimized poses in each local ba are used to construct the next layer's keyframe.

HaisenbergPeng, 2023-08-07:
okay, got it! Maybe I could try using the optimized poses of the bottom layer as factors.