hku-mars/HBA #6: Large RAM issue

wwtinwhu, 2023-06-16:
Hello, @samsdolphin . Great work! 

I tested with data of about 6k frames with 3, 4, 5 layer respectively. However, even with 5 layer setting, the RAM is exceed the max about 16GB. However, from your paper, the RAM cost is smaller than 5GB with 6k frames. What is the reason for this? 

[image]

samsdolphin, 2023-06-16:
For your scenario, I would recommend using 32GiB or even 64 GiB RAM computer, which can save the total computation time. In our early development stage, we calculate each layer separately, and we need to save and read each layer's pcds again and again. The local BA, global BA and pose graph optimization functions are performed in different `launch` files. I thought it might be better to combine them together. Hope this helps.

closed 2023-06-23 by wwtinwhu

wwtinwhu, 2023-06-23:
Thanks again for @samsdolphin. The issue here is caused by putting pose.json in the pcd directory. After correcting this, the RAM used is around 4GB with my test data.

seajayshore, 2023-06-23:
That's good. Thanks for confirming. 

Do the results look "good"? Does the HBA optimisation work well with your Mid-70 data?

wwtinwhu, 2023-06-24:
@samsdolphin Maybe add a check for opening pose file in read_pose function :)

wwtinwhu, 2023-06-24:
@seajayshore Hello! I am testing it now! What's your testing scene? Maybe we can communicate with email, and you can find my email in my profile.