hku-mars/HBA #7: std::bad_alloc while launch hba.launch with kitty07

JACKLiuDay, 2023-06-20:
[image]
Hi, thank you for your great work about HBA. I read the paper and try to launch the hba project with kitty07 dataset. But when I launch the file, something wrong happened.

JACKLiuDay, 2023-06-21:
Hi，I found the reason. It looks like my notebook memory is full. When I run the launch file, memory reach 16G soon. Then report std::bad_alloc

samsdolphin, 2023-06-21:
Hi @JACKLiuDay, yes, this is an issue, as mentioned in #6. It should be able to work under 32GiB RAM.

JACKLiuDay, 2023-06-21:
> Hi @JACKLiuDay, yes, this is an issue, as mentioned in #6. It should be able to work under 32GiB RAM.

Hi，thank you for your quick reply. 16G×2 memory module has been ordered on jingdong. I will try HBA again.

JACKLiuDay, 2023-06-23:
@samsdolphin Hi, I just tried my notebook with 16G×2, and it still shows std::bad_alloc. My notebook now has 27.6G  memory. Maybe I need to buy more RAM?

samsdolphin, 2023-06-23:
Hi @JACKLiuDay, this is weird. I am testing it on my own laptop (see configuration below), and it works fine for me (with maximum 18G RAM consumption). Can you make sure you are using c++17 (in `CMakeLists.txt`)? Previously I used c++14/c++11, and it failed like this.
[image]

[image]

samsdolphin, 2023-06-23:
Hi @JACKLiuDay, can you check [this reply](https://github.com/hku-mars/HBA/issues/6#issuecomment-1604188647) as mentioned in issue #6? This error is similar to yours.

JACKLiuDay, 2023-06-24:
> Hi @JACKLiuDay, can you check [this reply](https://github.com/hku-mars/HBA/issues/6#issuecomment-1604188647) as mentioned in issue #6? This error is similar to yours.

Hi, I tried the reply in #6. It's not working for me. My pose file was put in the right path and I checked it. 
The cmake was set c++17 and it did not work. My laptop  RAM is the same as yours. I have 27.6G RAM now. But when I launch kitti07, htop shows the RAM surging to 27G fast. Mabybe my CPU not working for this project? My CPU is AMD 5800H.

samsdolphin, 2023-06-24:
> Hi, I tried the reply in #6. It's not working for me. My pose file was put in the right path and I checked it. The cmake was set c++17 and it did not work. My laptop RAM is the same as yours. I have [...]

Ok, @JACKLiuDay, this is something I haven't been concerned about. You could try it on an Intel laptop. 0.0

wwtinwhu, 2023-06-25:
It is strange that the terminal has no output. @samsdolphin @JACKLiuDay You can add some print function in the construction of hba before the `printf(''HBA init done'') ` to see which part cause that issue. May help. 

[image]

JACKLiuDay, 2023-06-25:
> It is strange that the terminal has no output. @samsdolphin @JACKLiuDay You can add some print function in the construction of hba before the `printf(''HBA init done'') ` to see which part cause that issue. May help.
> 
> [image]

Hi, thank you for your advice. I will try it.

JACKLiuDay, 2023-06-25:
@samsdolphin @wwtinwhu Hi, I found the reason. But some new problem happeded. The function inf mypcl::readpose,   while(!file.eof()) became infinite loop. I changed the function. The HBA init successed. But the pcd load error happened. Some pcd file can not load. 
  std::vector<pose> read_pose(std::string filename,
                              Eigen::Quaterniond qe = Eigen::Quaterniond(1, 0, 0, 0),
                              Eigen::Vector3d te = Eigen::Vector3d(0, 0, 0))
  {
    std::vector<pose> pose_vec;
    std::fstream file;
    file.open(filename);
    double tx, ty, tz, w, x, y, z;
    while(!file.eof())
    {
      file >> tx >> ty >> tz >> w >> x >> y >> z;
      Eigen::Quaterniond q(w, x, y, z);
      Eigen::Vector3d t(tx, ty, tz);
      pose_vec.push_back(pose(qe * q, qe * t + te));
    }
    file.close();
    return pose_vec;
  }

samsdolphin, 2023-06-25:
> @samsdolphin @wwtinwhu Hi, I found the reason. But some new problem happeded. The function in mypcl::readpose, while(!file.eof()) became infinite loop on my computer. I changed the function. The HBA init successed. But the pcd load error happened. Some pcd file can not load. [image]

Hi @JACKLiuDay, thanks for your feedback and @wwtinwhu's suggestion. I guess the AMD processor caused these issues 0.0

JACKLiuDay, 2023-06-26:
> > @samsdolphin @wwtinwhu Hi, I found the reason. But some new problem happeded. The function in mypcl::readpose, while(!file.eof()) became infinite loop on my computer. I changed the function. The HBA init successed. But the pcd load error happened. Some pcd file can not load. [image]
> 
> Hi @JACKLiuDay, thanks for your feedback and @wwtinwhu's suggestion. I guess the AMD processor caused these issues 0.0 since I only tested on the Intel computer.

Hi, I found the reason finally. My data path is wrong. It nothing to do with CPU. HBA can run on my notebook successfully. And the result is good for the pointcloud. 
Excellent work!

closed 2023-06-26 by JACKLiuDay

wwtinwhu, 2023-06-26:
@JACKLiuDay 解决就好！

Gaaaavin, 2024-02-27:
@JACKLiuDay Hi, I have the same problem you met. I checked the file path and it is correct. I'm using the provided kitti07 dataset. Do you have any idea on how to debug?