hku-mars/HBA #8: There are only two layers in factor graph

jkff00, 2023-06-20:
hello!
          The number of pose graph optimization layers seems to be only the top and bottom layers, and the middle layer does not add factor maps to participate in optimization(upper_pose && init_pose in pose_graph_optimization())

          And I reproduced the park data in the paper, and the ground did improve, but as indicated by the white box, some areas showed degradation
[image]

samsdolphin, 2023-06-20:
Hi @jkff00 , for your first question, I refer you to #5. For the second question, the area you emphasized is a building surface, which indeed should not be converged. Please see the images below. 
[image]
[image]

jkff00, 2023-06-20:
Thank you for your reply. I understand.

closed 2023-06-20 by samsdolphin