hku-mars/HBA #9: PointCloud in visualization wrong

JACKLiuDay, 2023-06-26:
[image]
Hi, I got some new error with my own dataset. I run the LIO-SAM and save some pcd and pose. But when I launch HBA with this dataset. The visualization shows the pose correctly. But the pointcloud is wrong. The lidar is ouster-0-64U. Could you give me some advice? The transformation matrix in LIO-SAM pose graph need to be changed to fit HBA?

wwtinwhu, 2023-06-26:
@JACKLiuDay The pose format for HBA is x,y,z,qw,qx,qy,qz. You may check whether LIO-SAM pose format is consistent with that.

samsdolphin, 2023-06-26:
@JACKLiuDay, the point cloud should be in the `imu` coordinate system so that you can transform them into the `world` coordinate system. I guess the point cloud you used was in the `world` coordinate system. Hope this helps.

JACKLiuDay, 2023-06-26:
> @JACKLiuDay The pose format for HBA is x,y,z,qw,qx,qy,qz. You may check whether LIO-SAM pose format is consistent with that.

Hi, thank you for your advice. I will check it in my code.

JACKLiuDay, 2023-06-29:
> @JACKLiuDay, the point cloud should be in the `imu` coordinate system so that you can transform them into the `world` coordinate system. I guess the point cloud you used was in the `world` coordinate system. Hope this helps.

Hi, bro. I tried to transform point cloud into local frame and it worked. HBA works good. I am tring to use HBA for more dataset. Thank you for your great job.

closed 2023-06-29 by JACKLiuDay

pauldeee, 2023-08-19:
Hi @JACKLiuDay,

Is it possible for you to share your solution?

lvpengxiang, 2024-07-07:
I also encountered the same problem as you. Could you please tell me how you modified the code of liosam?

Silentbarber, 2024-12-14:
> 我也遇到了和你一样的问题。您能告诉我您是如何修改 liosam 的代码的吗？

我也遇到了一样的问题，pose.json里的位姿究竟是fastlio的什么位姿呢？是odometry吗？还是lidar frame的位姿

JACKLiuDay, 2024-12-16:
点云保存在 imu 坐标系下

***@***.***
***@***.***

Silentbarber, 2024-12-16:
> 点云保存在 imu 坐标系下 ***@***.*** ***
谢谢，其实就是fastlio发布出来的odometry对吧