hku-mars/HBA #10: 多次运行出现异常退出

MengHuiLei, 2023-06-27:
您好，万分感谢您开源这么好的算法！！！！ 膜拜大佬。。。。

最近在测试中遇到3个问题，向您咨询一下。@samsdolphin
1. 多次运行以后出现异常退出，如图：（备注：如果重新采用未优化以前的文件，）
[image]

2. 出现多处轨迹断点，或者说分层。建图时，这个位置是平滑过度的。
[image]

3. 部分区域的有点模糊
[image]

samsdolphin, 2023-06-28:
@MengHuiLei，您好，感谢您对我们工作的认可和关注。关于第一个问题，我这边暂时没有遇到类似的情况，如果方便可否请您共享一下测试数据，我这边也运行一下看看问题。关于问题2和3，根据我的经验，这个和BA参数相关，对于这种场景，可以试试初始voxel_size=1m、点云dowmsample_size=0.05、eigen_ratio=0.05、同时也令global_ba中的eigen_ratio=0.05。如果效果仍然没有改善，还有可能是遇到了退化场景，目前版本的hba没有imu，无法应对退化问题0.0。

MengHuiLei, 2023-06-29:
您好：
       非常感谢您的回复，今天我又把文件测试了一次。

抄送: "爱老婆  ***@***.******@***.***&gt;;

MengHuiLei, 2023-06-29:
您好：
       非常感谢您的回复，今天我又把文件测试了一次。
       问题1：我测试大概10次，出现问题； 问题2、3：只修改voxel_size=1m、dowmsample_size=0.05，效果感觉更差。（global_ba中的eigen_ratio=0.05 未找到）
       针对于退化问题，测试场景中特征较为明显，感觉应该不是退化的问题。（注：原始轨迹较为平滑，未出现断点问题。）

       文件链接：https://cowtransfer.com/s/f2e6ae09c27e46  提取码： bpp23b
       文件描述：
               1、launch文件中设置pcd_name_fill_num = 0，data_path也需要更改哈！！
               2、有2个pose文件，pose.json为提出问题的位姿文件，pose_ori.json为建图生成的原始文件。

       期待您的回复。。。。。。。

抄送: "爱老婆  ***@***.******@***.***&gt;;

samsdolphin, 2023-06-29:
@MengHuiLei，我试了一下你的数据，确实很有挑战性。我的参数是这样设置的：voxel_size在local_ba的时候为1m，在global_ba的时候为4m，然后eigen_ratio=0.1，点云dowmsample_size=0.05。迭代2-3次基本上大的地面分层都能合并。不过这个数据雷达每帧的点都不是很多，加上场景中有很多树丛，尤其是最后一个下坡的地方，对雷达BA来说是一个挑战。
[image]

samsdolphin, 2023-06-29:
[pose.txt](https://github.com/hku-mars/HBA/files/11907392/pose.txt)

MengHuiLei, 2023-06-30:
非常激动这么快就收到您的回复，通过测试约3次以后就很棒。（多说一句，再多次优化就会出问题）
针对您说的点少的问题，选择velodyne32线激光雷达，绕十字路口跑一圈。
从轨迹上看，位姿文件和点云文件都没有问题，但是一优化，结果就乱了。（尝试更改参数voxel_size、eigen_ratio、downsample_size，未能起效）
（注：初始姿态角未与map X轴重合，有一定夹角；起始位置点也不在零点；rviz看点云时，缩放一下就可以看到点云哈！！）
https://cowtransfer.com/s/3217cebd76a04f 提取码： t9d2eq
[image]

samsdolphin, 2023-06-30:
@MengHuiLei，这个我有空再看吧😄，你也可以试试调整不同的参数。

samsdolphin, 2023-07-02:
@MengHuiLei，你这个pcd是不是word系，没转换到imu系，我rviz看点云全是糊的...
[image]

MengHuiLei, 2023-07-03:
您好，很开心您能这么快做出回应！！！！ 
           1、先回答您的疑惑：确定已经转到imu系；
           2、问题出在：初始姿态角（rtk朝向）与map X轴的不重合上，测试50、30、10度均不能成功。即只有0度时，能够优化。
对于初始姿态角的问题，还在找解决方案。。。。

samsdolphin, 2023-07-03:
> 您好，很开心您能这么快做出回应！！！！ 1、先回答您的疑惑：确定已经转到imu系； 2、问题出在：初始姿态角（rtk朝向）与map X轴的不重合上，测试50、30、10度均不能成功。即只有0度时，能够优化。 对于初始姿态角的问题，还在找解决方案。。。。

我想知道你共享的点云和pose，在你那边用`visualize.launch`是不是也是糊的，如果是的话，那这个hba解决不了。

MengHuiLei, 2023-07-03:
实在对不住，是我个人原因（优化过位姿的文件上传了），浪费您的时间，万分抱歉。。。
https://cowtransfer.com/s/6a26bc467cb04f 读取码： 7h3y5x 
下面是visualize.launch 采用初始位姿看到的效果图
（注：1. rviz需要放缩一下，基准尺寸不在零点；2.解压以后，位姿和点云文件在名为32的文件夹中）
[image]

samsdolphin, 2023-07-03:
@MengHuiLei，首先你的pose位姿是不准确的...从侧面看，地面是糊的，不过可以用hba来优化这件事，local和globa ba的参数设置是一样的，voxel_size=2m、downsample_size=0.05m，reject_ratio=0.05，eigen_ratio=0.1。因为你的初始位姿坐标系很怪，所以你得改一下`mypcl.hpp`的`write_pose`函数，把`q0`置为单位阵，`t0`置为0，优化一次地面就变平了（[pose.txt](https://github.com/hku-mars/HBA/files/11937627/pose.txt)）。

[image]

[image]

MengHuiLei, 2023-07-04:
万分感谢。。。。。啥也不说了，请受小弟一拜  ！！！！
祝大佬科研之路，一路发发发SCI。。。。哈哈 ！！！
多说一句，期待新作。。

gongjun136, 2023-10-07:
> 您好，万分感谢您开源这么好的算法！！！！ 膜拜大佬。。。。
> 
> 最近在测试中遇到3个问题，向您咨询一下。@samsdolphin
> 
> 1. 多次运行以后出现异常退出，如图：（备注：如果重新采用未优化以前的文件，可正常运行）
>    [image]
> 2. 出现多处轨迹断点，或者说分层。建图时，这个位置是平滑过度的。
>    [image]
> 3. 部分区域的有点模糊
>    [image]
你好，我运行自己的数据也出现了异常，而且只运行了一次就出现了，不知道你是否已找到原因：
[image]

MengHuiLei, 2023-11-22:
您好：
    非常感谢您开源这么好的算法，我使用时效果很棒，给您点赞！！              
    我最近测试一个我采集的数据，无论如何都无法优化，下面我附上优化terminal输出和效果图：
    另外，我调试global_ba、local_ba中参数voxel_size=0.5、0.1、2、4m、downsample_size=0.05、001m，reject_ratio=0.05、0.1，eigen_ratio=0.05、0.1；均未能起作用。
    您空闲时，能帮忙看一下吗？  pose和pcd文件在附件中。
    期待您的回复。。。

抄送: "爱老婆  ***@***.******@***.***&gt;;

 

 

 

 

从QQ邮箱发来的超大附件

pcd+pose.zip (236.21M, 2023年12月22日 17:54 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=2037326641f0089229cf0bf11163064c19430702545301014c015300004e045558521f040005564e580756020f57055200560551376f3413025319165810514d1b5e42660a&t=exs_ftn_download&code=a72f7c4c

MengHuiLei, 2023-11-22:
不好意思哈！！ 回复晚了，最近在忙项目。
这个问题说实话，我没有具体定位到，但是大概率是位姿的问题。--- 我重新录制位姿就没问题了。

抄送: "爱老婆  ***@***.***&gt;;"State ***@***.***&gt;;

  
您好，万分感谢您开源这么好的算法！！！！ 膜拜大佬。。。。

MengHuiLei, 2023-11-23:
您好：非常感谢您开源这么好的算法，我使用时效果很棒，给您点赞！
补发优化terminal输出和效果图;
[image]
[image]

> 您好： 非常感谢您开源这么好的算法，我使用时效果很棒，给您点赞！！ 我最近测试一个我采集的数据，无论如何都无法优化，下面我附上优化terminal输出和效果图： 另外，我调试global_ba、local_ba中参数voxel_size=0.5、0.1、2、4m、downsample_size=0.05、001m，reject_ratio=0.05、0.1，eigen_ratio=0.05、0.1；均未能起作用。 您空闲时，能帮忙看一下吗？&nbsp; pose和pcd文件在附件中。 期待您的回复。。。 从QQ邮箱发来的超大附件 pcd+pose.zip (236.21M, 2023年12月22日 17:54 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=2037326641f0089229cf0bf11163064c19430702545301014c015300004e045558521f040005564e580756020f57055200560551376f3413025319165810514d1b5e42660a&t=exs_ftn_download&code=a72f7c4c

MengHuiLei, 2023-11-23:
> 您好： 非常感谢您开源这么好的算法，我使用时效果很棒，给您点赞！！ 我最近测试一个我采集的数据，无论如何都无法优化，下面我附上优化terminal输出和效果图： [...]

多说一句：采用纯fast_lio轨迹进行优化，参数比较随意，怎么都可以优化，而且效果都很棒；本轨迹为RTK优化一次，怎么调参数就感觉没动似的。

MengHuiLei, 2023-11-23:
> 您好： 非常感谢您开源这么好的算法，我使用时效果很棒，给您点赞！！ 我最近测试一个我采集的数据，无论如何都无法优化，下面我附上优化terminal输出和效果图： 另外，我调试global_ba、local_ba中参数voxel_size=0.5、0.1、2、4m、downsample_size=0.05、001m，reject_ratio=0.05、0.1，eigen_ratio=0.05、0.1；均未能起作用。 您空闲时，能帮忙看一下吗？&nbsp; pose和pcd文件在附件中。 期待您的回复。。。 https://cowtransfer.com/s/1751567b8fae42 点击链接查看 [ pose+pcd.zip ] ，或访问奶牛快传 cowtransfer.com 输入传输口令 xsr4v6 查看；

备注：更新位姿和pcd文件：
https://cowtransfer.com/s/1751567b8fae42 点击链接查看 [ pose+pcd.zip ] ，或访问奶牛快传 cowtransfer.com 输入传输口令 xsr4v6 查看；

samsdolphin, 2023-11-23:
@MengHuiLei，我看了一下你的点云，我的建议是加一个回环检测，首尾应该能检测到回环，然后再用hba优化。

MengHuiLei, 2023-11-24:
> @MengHuiLei，我看了一下你的点云，我的建议是加一个回环检测，首尾应该能检测到回环，然后再用hba优化。

懂您的意思，我是这样考虑的，高程差为3米左右，算法应该可以优化过来；并且我用您的算法，差6米的时候也能稳定优化过来。
重点，重点，重点：采用纯fast_lio轨迹进行优化，参数比较随意，怎么都可以优化，而且效果都很棒。
（备注：优化不过来的轨迹：RTK中的xy优化过一次的轨迹）

不知道，我提供的这些信息是否有帮助定位到问题。。。

MengHuiLei, 2023-11-24:
补充：这次的位姿文件，把yaw也更新了，就可以优化
https://cowtransfer.com/s/96098b2ce86f43 点击链接查看 [ pcd+pose.zip ] ，或访问奶牛快传 cowtransfer.com 输入传输口令 et4wma 查看；

closed 2023-11-24 by MengHuiLei