hku-mars/HBA #14: 优化多次基本没什么变化

SlamSmallFish, 2023-07-20:
您好，请问我运行多次hba.launch，但是优化出来的结果几乎没什么变化，无法回到同一地方，无法像您在github上展示的kitti00那样最后回到起点，可以提供一些建议吗？
[image]
[image]

JACKLiuDay, 2023-07-26:
> @SlamSmallFish 你好，我的建议是先做一个回环检测+位姿图优化，再用hba优化剩余分层。你的这个场景只有首尾这里有匹配关系，如果voxel_size不够大，没有把图二分层的地面放到一个voxel内，其实位姿就没有被优化。

Hi, I tried something new on my campus with HBA. This environment only has a closed loop at the beginning and end. And the STD project works not so well. Do you have some advice.  How to form an effective closed loop for an environment that only has a closed loop at the beginning and the end?  Moreover, this environment is still a large symmetrical circle, and many closed-loop detections have failed.  https://github.com/hku-mars/STD/issues/22

samsdolphin, 2023-07-26:
Hi @JACKLiuDay, we have encountered this issue before. My suggestion is to manually select the keyframe and perform the ICP or whatever to determine the relative transformation and add it to the pgo.

JACKLiuDay, 2023-07-27:
> Hi @JACKLiuDay, we have encountered this issue before. My suggestion is to manually select the keyframe and perform the ICP or whatever to determine the relative transformation and add it to the pgo.

Hi, thank you for your advice. I will try it.

samsdolphin, 2023-07-27:
@SlamSmallFish any updates on this issue?

closed 2023-07-28 by samsdolphin