hku-mars/HBA #15: PointClouds are not visualized properly

pauldeee, 2023-08-04:


samsdolphin, 2023-08-07:
Hi @pauldeee, I think this is the LIO-SAM issue. There is some extrinsic between the LiDAR and the imu in the LIO-SAM configuration. Please make sure you have the point cloud and the corresponding poses in the same frame, i.e., both in the LiDAR body frame or the imu frame.

pauldeee, 2023-08-07:
Hi @samsdolphin,

I figured it out. The documentation states the pose file should have the format `tx ty tz qw qx qy qz`.

[image]

When I switch my file to have the poses in the format `tx ty tz qx qy qz qw` it works.

Thanks!

pauldeee, 2023-08-07:
Sorry, I may have spoken too soon.

It seems while the change I made allows the `visualize.launch` to work correctly `hba.launch` does not produce the expected results...

pauldeee, 2023-08-09:
Hi @samsdolphin,

Here is a link to some test data: https://drive.google.com/drive/folders/1tn7ZFGnVPS39j8jd3yWux9ykgKQkKUXV?usp=sharing

This works fine with the `visualize.launch`. Could it be a TF issue that's causing it not to work with the `hba.launch` even though it works fine with the `visualize.launch`?

Thanks!

samsdolphin, 2023-08-13:
Hi @pauldeee, I checked your data and I found only 97 poses (pcds). If this is your total number of point clouds, I would suggest you use the [BALM](https://github.com/hku-mars/BALM) directly as HBA is designed to handle large-scale point cloud mapping, e.g, 1K or 10K or more poses.

closed 2023-08-25 by samsdolphin

lvpengxiang, 2024-07-08:
I also encountered the same problem as you. Could you please tell me how you modified the code of liosam?