hku-mars/HBA #19: HBA优化问题

gongjun136, 2023-11-02:
作者你好，为了测试室内的效果，我没有再使用多层楼数据，主要最近实验发现里程计在走廊退化严重，于是录制了地下车库数据做测试。我是用gtsam进行pgo后的里程计作为输入，发现HBA后变化不大。数据起始点和终点是同一个位置，通过看这个位置的地图就可以看到仍然有许多分层，我使用的是其域的16线激光扫描仪手持设备，HBA优化后分层如下：
[image]

如下是数据，作者可以帮忙看一下吗：
https://cowtransfer.com/s/68f9c796a6d248 点击链接查看 [ hba_data.rar ] ，或访问奶牛快传 cowtransfer.com 输入传输口令 ufnwq1 查看；

samsdolphin, 2023-11-15:
> 作者你好，为了测试室内的效果，我没有再使用多层楼数据，主要最近实验发现里程计在走廊退化严重，于是录制了地下车库数据做测试。我是用gtsam进行pgo后的里程计作为输入，发现HBA后变化不大。数据起始点和终点是同一个位置，通过看这个位置的地图就可以看到仍然有许多分层，我使用的是其域的16线激光扫描仪手持设备(20hz)，HBA优化后分层如下： [image]
> 
> 如下是数据，作者可以帮忙看一下吗： https://cowtransfer.com/s/68f9c796a6d248 点击链接查看 [ hba_data.rar ] ，或访问奶牛快传 cowtransfer.com 输入传输口令 ufnwq1 查看； 这是设备的样子： [image]

@gongjun136 你好，抱歉现在才看到回复，前段时间在忙毕业。你的问题解决了吗？我试着查看你的文件，发现已经过期了。

gongjun136, 2023-11-15:
@samsdolphin 还没有，如下是数据集链接：
https://cowtransfer.com/s/07842eaf85a444 点击链接查看 [ hba_data.rar ] ，或访问奶牛快传 cowtransfer.com 输入传输口令 3fcqzb 查看；
我测试了BALM，用前50帧进行优化，发现BALM可以优化图中的角度偏差，但是使用HBA，在参数相同情况下却并没有优化过来。如下是对应的位姿文件：
[alidarPose.csv](https://github.com/hku-mars/HBA/files/13361397/alidarPose.csv)
[image]

samsdolphin, 2023-11-15:
我看了一下你的数据，除了最开始几十帧分层，后面点云看起来没啥问题，直接去除前几帧应该就行了吧？
另外看你最下面ba后的图，效果也很好，那直接把对应参数用在hba上ok吗？

gongjun136, 2023-11-15:
@samsdolphin 我代码中最开始是做了初始化的，这些数据按理都是初始化后可用的，但是在这个数据集中最开始时的计算因为约束不够产生了较大误差，需要多帧的约束来纠正回来，这不就是BA的优势吗?并且从初始化后的所以帧都会用来更新全局体素地图，之后可能我还想做一个局部BA后更新地图。  我使用相同的参数(WIN_SIZE=50-60,gap=20-30,eigen_ratio=0.0625,voxel_size=2-4)放在hba上了，但是并没有优化过来

gongjun136, 2023-11-15:
@samsdolphin 对了，还想请问下作者关于论文中提到的地图质量评价的指标MME的代码是否有开源，我想评价一下BA前后的地图质量，我看原论文好像没提到github链接

samsdolphin, 2023-11-23:
> @samsdolphin 对了，还想请问下作者关于论文中提到的地图质量评价的指标MME的代码是否有开源，我想评价一下BA前后的地图质量，我看原论文好像没提到github链接

@gongjun136，我找一下，开源肯定没问题

gongjun136, 2023-11-24:
@samsdolphin 十分感谢，如果开源了还请给个github地址

aditdoshi333, 2024-01-22:
@gongjun136 你是如何解决这个漂移问题的？我最近看到xgrids转移到了livox mid 360。你们使用了什么样的校正或者捆绑调整算法？

samsdolphin, 2024-01-26:
> @samsdolphin 十分感谢，如果开源了还请给个github地址

@gongjun136，你好，这个计算MME的代码已经更新，可以试试～

closed 2024-01-26 by samsdolphin