hku-mars/HBA #20: 关于地图精细度的问题

ZYXiao-g, 2023-11-07:
作者您好，感谢提出这么惊艳的算法，关于地图的精细度有个问题想请教下作者。
下面第一个图是用官方数据，rviz里面透明度alpha默认0.1时的效果，效果非常好
[image]
但是把透明度切换成1.0的时候就会发现墙面和树其实非常的厚存在很多杂点。
[image]

最终保存成地图的效果是透明度1.0时存在大量杂点、墙面、树很厚时的效果，这种杂点和厚度主要是激光雷达本身的杂点造成的还是姿态不准确造成？ 请问作者有什么建议或者办法解决这种问题吗

ZYXiao-g, 2023-11-07:
[image]

samsdolphin, 2023-11-11:
@ZYXiao-g 你好，感谢使用我们的算法。关于你提到的杂点问题，这个主要是雷达测量造成的。过多的噪点确实会影响到位姿优化的精度，不过少量的噪点对我们方法的影响很小。去除这些噪点可能需要一些滤波算法。

closed 2024-01-26 by samsdolphin