hku-mars/HBA #21: to undistort the points or not

CharlieV5, 2023-11-19:
Hi, I noticed that there is a variable defined as `undis_t`, which is used to record the time taken to remove distortion from the point cloud. However, it is not being used in practice, so `undis_t` is always equal to 0. I would like to ask if it is necessary to remove distortion from the point cloud at each iteration of the BA algorithm. In your algorithm, is the input `pcd` point cloud the original or the one with distortion removed? I guess it is the one with distortion removed because the output of each iteration does not change the coordinates of the points in `pcd`, but only updates the pose of each frame of the point cloud. In theory, should the original point cloud be input and the pose of each frame be iteratively optimized in the BA algorithm while removing distortion from the point cloud? After all, the pose of the point cloud frame changes after each iteration, and removing distortion actually depends on these poses.

samsdolphin, 2023-11-19:
Hi @CharlieV5 , you are right, you can just delete this variable. The input `pcd` is already undistorted.

CharlieV5, 2023-11-19:
> Hi @CharlieV5 , you are right, you can just delete this variable. The input `pcd` is already undistorted.

Thank you! And I want discuss  the second topic, should the BA algorithm regenerate the undistorted point cloud, since the poses will be updated after each iteration and the distortion removal relies on the poses of frames?

closed 2024-01-26 by samsdolphin