hku-mars/HBA #22: 非重合平面之间变为重合

gongjun136, 2023-11-28:
作者你好，不好意思又来提问了，我记得BA算法会对体素内的平面进行约束优化，但是我发现部分不是同一个平面的被优化后变为了重合的平面，如下图所示：
[image]
[image]
这就导致了其他部分的分层：
[image]
[image]

gongjun136, 2023-12-11:
@samsdolphin 好的，感谢回复，我会去试试更严格的参数，作者的HBA2预计什么时候发布啊，想看一下最新的代码思路

samsdolphin, 2024-01-26:
@gongjun136，快了，刚刚毕业，目前在写文章～

closed 2024-01-26 by samsdolphin

gongjun136, 2025-04-22:
> [@gongjun136](https://github.com/gongjun136)，快了，刚刚毕业，目前在写文章～

作者你好，HBA2是否已经发表了？想参考一下优化后的BA效果