hku-mars/HBA #29: 运行自己的数据，报错，求指导。

JiangWeiHn, 2024-03-25:
我采用lio-sam运行自己的数据集，生成了pcd文件和pose.json.运行程序后，报了以下错误，不明白为什么需要这么多其他的pcd文件，请指导下，谢谢！！
.. logging to /home/slamer/.ros/log/39f31bc0-ea4e-11ee-82a4-c38df60e1741/roslaunch-slamer-8607.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://slamer:42465/

SUMMARY
========

PARAMETERS
 * /hba/data_path: /home/slamer/work...
 * /hba/pcd_name_fill_num: 0
 * /hba/thread_num: 16
 * /hba/total_layer_num: 3
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    hba (hba/hba)

auto-starting new master
process[master]: started with pid [8622]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 39f31bc0-ea4e-11ee-82a4-c38df60e1741
process[rosout-1]: started with pid [8639]
started core service [/rosout]
process[hba-2]: started with pid [8642]
init parameter:
layer_num 1 | thread_num 16 | pose_size 5519 | max_iter 10 | part_length 69 | gap_num 1101 | last_win_size 9 | left_gap_num 67 | tail 4 | left_size 679 | left_h_size 343 | j_upper 68 | downsample_size 0.100000 | voxel_size 4.000000 | eigen_ratio 0.100000 | reject_ratio 0.050000
hessian_size: 49626
init parameter:
layer_num 2 | thread_num 16 | pose_size 1103 | max_iter 10 | part_length 14 | gap_num 218 | last_win_size 8 | left_gap_num 9 | tail 3 | left_size 98 | left_h_size 52 | j_upper 10 | downsample_size 0.100000 | voxel_size 4.000000 | eigen_ratio 0.100000 | reject_ratio 0.050000
hessian_size: 9883
init parameter:
layer_num 3 | thread_num 15 | pose_size 220 | max_iter 10 | part_length 3 | gap_num 42 | last_win_size 5 | left_gap_num 1 | tail 0 | left_size 10 | left_h_size 9 | j_upper 1 | downsample_size 0.100000 | voxel_size 4.000000 | eigen_ratio 0.100000 | reject_ratio 0.050000
hessian_size: 24090
HBA init done!
---------------------
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/345.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/346.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/347.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/348.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/349.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/350.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/351.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1035.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1036.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1037.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/352.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1380.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1038.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/353.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1039.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1381.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1382.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/690.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/354.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1383.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1384.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1725.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1040.pcd'.
[pcl::PCDReader::read] Could not find file '/home/slamer/workspace/HBA-main/pcd/1385.pcd'.
[hba-2] process has died [pid 8642, exit code -11, cmd /home/slamer/workspace/HBA-main/devel/lib/hba/hba __name:=hba __log:=/home/slamer/.ros/log/39f31bc0-ea4e-11ee-82a4-c38df60e1741/hba-2.log].
log file: /home/slamer/.ros/log/39f31bc0-ea4e-11ee-82a4-c38df60e1741/hba-2*.log

tx-lidar, 2024-04-01:
Hello, how did you obtain this json file and are there any format requirements?

JiangWeiHn, 2024-04-01:
i have fixed this problem,when run the bagfile,need save every keyframe lidar point to *pcd,the pose.json‘s format is tx ty tz qw qx qy qz .

closed 2024-04-01 by JiangWeiHn

lvpengxiang, 2024-06-28:
Could you please tell me how you modified LIO-SAM to output each PCD along with the corresponding pose? I have tried several times without success. Could you kindly share the relevant code with me? Thank you.