hku-mars/HBA #33: Questions about HBA results

aksrb1030, 2024-04-11:
Hi.
Thanks for your great work!
I tried to my own dataset but result something wrong!
Sensor : Hesai XT32
```
max_iter = 30;
donwsample_size = 0.05;
voxel_szie = 2.5;
eigen_ratio = 0.05;
reject_ratio = 0.05;
```
Before, HBA
[image]

After, HBA
[image]

Please know how to soved!

samsdolphin, 2024-04-12:
Hi @aksrb1030, thanks for using our work. From your screenshot, I believe the initial trajectory is of good quality and the parameter setting is fine, I think it might be due to some configuration issues. Maybe you could share your data with me, so I could try it with my laptop?

aksrb1030, 2024-04-16:
Sure!! I sent you google drive link your email to liuxiyuan95@gmail.com

phucvinh-nguyen, 2024-04-22:
Hi @samsdolphin, i met the same problem, i realize that the x coordinate has changed sign after optimization. Please help me to solve it!

phucvinh-nguyen, 2024-04-22:
Can i send my data to your email - liuxiyuan95@gmail.com?

tempted111, 2024-05-10:
Hi.
Thanks for your great work!
I tried to my own dataset but i do not konw how to get tx,ty,yz,qw. Are their values obtained through the mean of the point cloud？

samsdolphin, 2024-05-20:
> Hi. Thanks for your great work! I tried to my own dataset but i do not konw how to get tx,ty,yz,qw. Are their values obtained through the mean of the point cloud？

The pose of each LiDAR scan can be obtained from a general LIO system, so for example, you can use fastlio to get it.

tempted111, 2024-05-26:
> > Hi. Thanks for your great work! I tried to my own dataset but i do not konw how to get tx,ty,yz,qw. Are their values obtained through the mean of the point cloud？
> 
> The pose of each LiDAR scan can be obtained from a general LIO system, so for example, you can use fastlio to get it.

Thanks, Thanks for your great work! Our data has a pose every 8ms, which generates over 8000 PCD files. Therefore, I want to merge 15 poses to form a pose for 200ms, The number of PCD file files will be reduced to about 1000. How should I merge these 15 poses? I tried to take the average of these 15 poses before, but the result was poor. Can you give me some suggestions? Thank you

tempted111, 2024-06-14:
Hi.
Thanks for your great work!
I tried to my own dataset but result something wrong!
Before HBA:
[image]
After HBA:
[image]
Please know how to soved!

tempted111, 2024-06-14:
I do not modify the parameter

chengwei0427, 2024-10-11:
> Hi. Thanks for your great work! I tried to my own dataset but result something wrong! Sensor : Hesai XT32
> 
> ```
> max_iter = 30;
> donwsample_size = 0.05;
> voxel_szie = 2.5;
> eigen_ratio = 0.05;
> reject_ratio = 0.05;
> ```
> 
> Before, HBA ![Screenshot from 2024-04-11 11-41-47](https://private-user-images.githubusercontent.com/18605999/321595236-6e664852-2d99-405f-b6f7-2546aebdc645.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg2NDE5MDUsIm5iZiI6MTcyODY0MTYwNSwicGF0aCI6Ii8xODYwNTk5OS8zMjE1OTUyMzYtNmU2NjQ4NTItMmQ5OS00MDVmLWI2ZjctMjU0NmFlYmRjNjQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDExVDEwMTMyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIwM2Y0ZDIyM2Q1MjkwMzZlMjlmYzNmYjFkNGMzODU0ZTQyY2YwYTc5MmQwNTQwMzllMTRlYTVmMWVmODM2NTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.uRNPk8vD-YzgzyBlF8TGG05UUzDqBbqQXFIwkB5rVHs)
> 
> After, HBA ![Screenshot from 2024-04-11 14-55-43](https://private-user-images.githubusercontent.com/18605999/321595213-6a6a15d5-01de-4946-bb6e-e40c04fd27eb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg2NDE5MDUsIm5iZiI6MTcyODY0MTYwNSwicGF0aCI6Ii8xODYwNTk5OS8zMjE1OTUyMTMtNmE2YTE1ZDUtMDFkZS00OTQ2LWJiNmUtZTQwYzA0ZmQyN2ViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDExVDEwMTMyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2MzBhNjFiMmYwNWY0ZjlkNzJjY2JlZDc3MjdkZDBkYzRiMDVkMzQzMThjNzZkYzZmZTI4NjI2NDk3NjRmYmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.f8SbFNsQY3xkm3K-psU-o90v2G8JRf3977-h4QM7yQs)
> 
> Please know how to soved!

Has the problem been solved? Is there any new progress on this issue?

chengwei0427, 2024-10-16:
Hi, @samsdolphin @aksrb1030 @tempted111 @phucvinh-nguyen 

There might be a bug in the write_pose function. I've submitted a [PR](https://github.com/hku-mars/HBA/pull/39/commits/9e2960fb2abd0f72ef9fa1638e5a521c4ec51a53) for it, and you can take a look at it for reference.

yuxuanzhou97, 2024-10-18:
> Hi, @samsdolphin @aksrb1030 @tempted111 @phucvinh-nguyen
> 
> There might be a bug in the write_pose function. I've submitted a [PR](https://github.com/hku-mars/HBA/pull/39/commits/9e2960fb2abd0f72ef9fa1638e5a521c4ec51a53) for it, and you can take a look at it for reference.

You save me!!!