hku-mars/HBA #38: Why does negative optimization occur on my dataset

Tartisan, 2024-09-12:
[image]
above is the initial effect

[image]
this is the effect after HBA
I don't know why

chengwei0427, 2024-10-16:
Hi， @Tartisan 
There might be a bug in the write_pose function. I've submitted a [PR](https://github.com/hku-mars/HBA/pull/39/commits/9e2960fb2abd0f72ef9fa1638e5a521c4ec51a53) for it, and you can take a look at it for reference.

G4419, 2024-11-07:
@chengwei0427 Yes, you are correct. The original write_pose function set first pose to Identity pose, which is wrong.